srl-freiburg/pedsim_ros #8: Pedestrian Walking Sim for Obstacle Avoidance

ktiwari9, 2015-08-05:
Hello there,
I am trying to make a sim for p3dx (Pioneer 3dx Robot) in RVIZ such that when going from point A to Goal position it has to avoid a moving pedestrian who may be walking across or just standing in its path so it has to re-plan its trajectory. Could you please tell me how can i use your animated_walking_man marker for this purpose ?
I would appreciate if you could reply at the earliest.

makokal, 2015-08-05:
The `animated_markers` simply provides a visualization of walking person in `rviz`. In pedsim these are the simulated people using the social force model.

Given your walking or standing person pose say `p = (x, y, vx, vy,....)`, you need to make a node that publishes the animated marker at every time step using such format.

```c++
  // walking person message (not complete)
        animated_marker_msgs::AnimatedMarker marker;
        marker.mesh_use_embedded_materials = true;
        marker.header.frame_id = "odom";
        marker.header.stamp = ros::Time();
        marker.ns = "some_namespace";
        marker.id = some_unique_marker_id;
        marker.type = animated_marker_msgs::AnimatedMarker::MESH_RESOURCE;
        marker.mesh_resource = "package://your_package/images/animated_walking_man.mesh";

        marker.pose.position.x = pose_x;
        marker.pose.position.y = pose_y;
        marker.action = 0;  // add or modify
        marker.scale.x = size_person_x;
        marker.scale.y = size_person_y;
        marker.scale.z = size_person_z;
       // ... additional marker attributes like color etc.

```
Then in rviz, you will be able to see a new marker type.

For the above animated marker to work, your just need the animated marker plugin package compiled in your namespace. You can take the walking people mesh resources from `pedsim_ros`.

ktiwari9, 2015-08-07:
Hi there,
So I did clone both animated_marker and ped_sim repo into my ~catkin_ws/src folder and I get an error because of find_package(catkin REQUIRED COMPONENTS ${PEDSIM_SIMULATOR_DEPENDENCIES}) in the cmakeLists file. I am using ros hydro and I have no idea how to fix this. I guess unless I can do catkin_make without errors I cannot use the AnimatedMarkers into my source code.
Please advise.

makokal, 2015-08-07:
This is probably because of the `cmake_modules` package which was only added in indigo.

You can see the `hydro` branch which does not involve this here 
[https://github.com/srl-freiburg/pedsim_ros/tree/hydro](https://github.com/srl-freiburg/pedsim_ros/tree/hydro)

From your already cloned version, do the following

```bash
git fetch
git checkout hydro

# the go ahead and compile
```

makokal, 2015-09-03:
I am closing this issue as I believe the last response solve it, otherwise we'll reopen

closed 2015-09-03 by makokal