srl-freiburg/pedsim_ros #10: Sensor Point Cloud Visualization Not Working

sfchik, 2016-06-03:
Hi,

Thanks for releasing such a useful pedestrian simulation package, it helps me a lot. Recently, I am trying out the pedsim_point_clouds package for simulation, but nothing can be seen in the rviz even after adding the point cloud topics. I tried to troubleshoot and found that the point cloud node is subscribing /spencer/sensors/odom, which I can't see any node publishing the topic. Hence, I suspect something wrong with my use case. May I know the correct step of using the pedsim_point_clouds package? Thank you.

makokal, 2016-06-03:
Hi,

Thanks for the feedback

The `pedsim_point_clouds ` needs the position of the robot. Its my mistake I set that to `/spencer/sensors/odom` which comes from our simulated odometry (`nav_msgs::Odometry`) which is not release with this package. 

Instead this should subscribe to `/pedsim/robot_position` published by the simulator which is a `geometry_msgs::PoseStamped` thing.

I will make the necessary change this tomorrow once am back to a real PC, but for now you can make a small temporary fix with by making it subscribe to the right topic.

sfchik, 2016-06-03:
Hi,

Thanks for your swift reply. Appreciate it. I will try it out and feedback to you after it. Thanks a lot :)

closed 2016-06-04 by makokal

sfchik, 2016-06-04:
Hi,

Thanks it is working. XD

commit 3aae6d33 (2016-06-04): Fixes #10 point cloud sensor data problem

- Fixes
- Now publishing robot position from the sim as odom message
- Point cloud sensor simulation now used the odom from the sim
- Make point cloud sensor code more generic wrt to msg tf frames.