srl-freiburg/pedsim_ros #15: Compilation problems

zeynab, 2016-10-04:
Hi,
I am trying to compile the pedsim_ros/pedsim_simulator and I get many problems with boost libraries, e.g.,  

```
from /home/talebpou/repos/svn/monarch/code/trunk/catkin_ws/src/pedsim_ros/pedsim_simulator/src/simulate_diff_drive_robot.cpp:1:
/usr/include/boost/function/function_template.hpp:1006:3: error:   initializing argument 1 of 'boost::function1<R, T1>::result_type boost::function1<R, T1>::operator()(T0) const [with R = void; T0 = boost::shared_ptr<const geometry_msgs::Twist_<std::allocator<void> > >; boost::function1<R, T1>::result_type = void]'
```
and many more.
I am using ROS hydro on ubuntu 12.04 and my boost version is: 1.48.0.2.
I have been unable to figure out what is wrong. 

Thanks a lot in advance for your help.
Zeynab

bp177, 2016-10-18:
i think this is just a boost version thing. Usually different ros versions support only a certain range of boost versions. From what I can tell, this code is targeted at indigo and kinetic

makokal, 2018-03-05:
The simulator is tested on indigo anf kinetic on 14.04 and 16.04 ubuntu versions. 
However, we do not see any reason it should not work in the ROS and ubuntu versions after 14.04 to today.

closed 2018-03-05 by makokal