srl-freiburg/pedsim_ros #17: Change max velocity of pedestrians?

pirobot, 2017-04-11:
Hello,

Thank you for this awesome simulator.  I was wondering if there is a configuration parameter that would enable me to change the maximum velocity for the pedestrians?  I am doing some computer vision tests on  detecting groups of moving people and it would be helpful to slow down the motion of the simulated people.

Thanks!

sfchik, 2017-04-11:
Hi pirobot,

What I did was I found the velocity parameter of the pedestrian under the file [your_workspace]/src/pedsim_ros/pedsim/src/ped_agent.cpp

under the code section:
    // assign random maximal speed in m/s
    // normal distribution (mean 1.2, std 0.26)
    normal_distribution<double> distribution ( 1.2, 0.06 );

You could try to change the value 1.2 to some other value and recompile the package. This does the job for me. Hopefully it works for you too.

Regards,
Chik
    vmax = distribution ( generator );

pirobot, 2017-04-11:
Hi Chik--many thanks for your quick reply.  Your suggestion works perfectly for me.  Thank you!

closed 2017-04-11 by pirobot