srl-freiburg/pedsim_ros #18: Documentation for scenario parameters?

pirobot, 2017-04-12:
Hello,

Is there documentation anywhere for the scenario parameters?  In particular, what are the precise meaning of the dx and dy parameters?

Thanks!

sfchik, 2017-04-12:
Hi Pirobot,

From my trial and testing, I found out the dx and dy is the position tolerance for the generation of pedestrian, lets say dx = 1 and dy = 1, if you want your pedestrian to be generated at x = 5 and y = 5, there will be +- 1 tolerance, so your pedestrian will be generated randomly within the tolerance. You can try to set them to 0, and you will find the pedestrian always generate accurately at where you want them to be. However, you can't set them to 0 if you wanted to spawn a group of pedestrian, it will return some kind of error.

Hope this help, and correct me if I am wrong.

Best Regards,
Chik

pirobot, 2017-04-12:
Thanks Chik.  I was hoping it had something to do with how agents grouped together--for example, walking side-by-side versus following each other.  But you might be right...

makokal, 2017-08-17:
Again sorry for a late response. The params `dx` and `dy` are the length and width extends on which a random position is chosen to spawn an agent. If they are zero, the agent is placed at the exact position, otherwise for a group of agents, they will be spawned in locations sampled from the 2D rectangle defined by these extents.

closed 2017-08-17 by makokal