srl-freiburg/pedsim_ros #21: Size of robot and pedestrians?

pirobot, 2017-08-16:
Does anyone know if the pedsim simulator assigns a size (e.g. diameter) to the robot or the pedestrians?  Or are they treated as points?  I ask because I cannot tell by looking at the RViz animations if the pedestrians are actually avoiding the robot or even each other since the animations appear to pass by each other very closely.

Thanks!
patrick

makokal, 2017-08-17:
Hi @pirobot , yes the size is set, see [Line 43 in the constructor](https://github.com/srl-freiburg/pedsim_ros/blob/master/pedsim/src/ped_agent.cpp#L43).

This is used in the social force computation. But the weight of other forces may overwhelm this component depending on the context.

pirobot, 2017-08-17:
Awesome--thanks @makokal!

closed 2017-08-17 by pirobot

RGring, 2018-10-18:
Unfortunately I can't access the link. Could you refresh it? Me too, I would need to reconfigure the size of the robot.
Thanks already!
Ronja

makokal, 2018-10-18:
@RGring  Here it is: https://github.com/srl-freiburg/pedsim_ros/blob/master/3rdparty/libpedsim/src/ped_agent.cpp#L34