srl-freiburg/pedsim_ros #22: Obstacles not avoided by agents

pirobot, 2017-09-01:
I am using the latest code from the master branch and it appears that the agents do not avoid obstacles as I thought they would, even if I increase the force_obstacle parameter to its max value of 50 and the force_wall parameter to its max value of 10.

I've attached a screen shot when running the simple_pedestrians.launch example after adding an obstacle to the social_contexts.xml file using the following line:

    <obstacle x1="14" y1="14" x2="17" y2="17"/>

You can see the obstacle in the screen shot and you can also see pedestrians all over it.   Also, changing the force_obstacle parameter using rqt_reconfigure seems to have no effect on where the agents pass over or through the obstacle. 

What do I need to do to make the agents avoid obstacles? 

Thanks!

[image]

framarbon, 2017-11-07:
I have noticed the same issue. Any solution?

tlind, 2017-11-07:
It's been a long time since I last used pedsim_ros. But I remember there was sometimes an issue about the grid being offset by half a cell. Can you make sure (via some experiments) that it is not just a grid alignment issue? The agents north of the obstacle seem to be lining up along a diagonal line, after all. Just a guess, though!

makokal, 2018-02-08:
This is probably just like you say grid being offset. Additionally remember that internally in the SFM obstacles are just lines. The thickness shown is just for visualization in terms of cells

closed 2018-09-03 by makokal

RGring, 2018-09-17:
Are the agents following a certain path planning algorithm? My agents get stuck in more complex environments (e.g. inner corners inside the room). Are there any possibilities to solve this?

[image]