srl-freiburg/pedsim_ros #23: Visualization doesn't compile

ksatyaki, 2017-11-16:
Hello guys, 
I'm Chitt from Orebro University.

I'm trying to compile the spencer_tracking_rviz_plugin on lunar. (Everything else compiles). 
In particular I get some scope resolution issues when the MOC is compiled. 
"pedsim_ros/visualization/spencer_tracking_rviz_plugin/CMakeFiles/spencer_tracking_rviz_plugin.dir/spencer_tracking_rviz_plugin_automoc.cpp.o"

A small snippet of the error should probably be enough to understand the issue. If this isn't enough, I can post the whole error. 

```
In file included from /home/chitt/workspace/ros_ws/build/pedsim_ros/visualization/spencer_tracking_rviz_plugin/spencer_tracking_rviz_plugin_automoc.cpp:6:0:
/home/chitt/workspace/ros_ws/build/pedsim_ros/visualization/spencer_tracking_rviz_plugin/spencer_tracking_rviz_plugin_automoc.dir/moc_social_activit_CTUXWLMTHHTXP4.cpp:68:27: error: ‘ros::message_traits::spencer_tracking_rviz_plugin’ has not been declared
 void ros::message_traits::spencer_tracking_rviz_plugin::SocialActivitiesDisplay::qt_static_metacall(QObject *_o, QMetaObject::Call _c, int _id, void **_a)
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/chitt/workspace/ros_ws/build/pedsim_ros/visualization/spencer_tracking_rviz_plugin/spencer_tracking_rviz_plugin_automoc.dir/moc_social_activit_CTUXWLMTHHTXP4.cpp: In function ‘void qt_static_metacall(QObject*, QMetaObject::Call, int, void**)’:
/home/chitt/workspace/ros_ws/build/pedsim_ros/visualization/spencer_tracking_rviz_plugin/spencer_tracking_rviz_plugin_automoc.dir/moc_social_activit_CTUXWLMTHHTXP4.cpp:71:9: error: ‘SocialActivitiesDisplay’ was not declared in this scope
         SocialActivitiesDisplay *_t = static_cast<SocialActivitiesDisplay *>(_o);
```

Could you help me out on this?

The current version I'm compiling is on my fork of the repository @ [github.com/ksatyaki/pedsim_ros](https://github.com/ksatyaki/pedsim_ros)

makokal, 2018-02-08:
Hey, sorry I am probably late at this, but we have updated the viz to now work on `indigo` and `kinetic`

ksatyaki, 2018-03-05:
Sweet! Compiles in no time. Thanks a lot!

closed 2018-03-05 by ksatyaki