srl-freiburg/pedsim_ros #24: No Occluded Agents

framarbon, 2017-11-23:
Hi there,
First of all, thanks for sharing this great tool for crowd simulation.

I am trying to simulate a robot navigating through a human crowd using pedsim, for that I ma using the simulated simulated pedsim_point_clouds. However it seems that both the pedsim simulator and the point cloud sensor does not take into account whether an agent might be occluded.

Is there any workaround or anything that I am missing to make it work?

Best regards,
Framarbon

makokal, 2017-11-23:
Hi @framarbon You are right, the point clouds do not take that into account. You can add an additional filter say based on distance and orientation to simulate that. We had a hook to gazebo to do this more naturally with ray casting but I haven't looked into that for a long time.

tlind, 2017-11-25:
You could use the following node from the SPENCER people tracking framework as a starting point for agent-to-agent occlusion, though it does not take into account occlusion by the static environment: [simulate_occluded_detections_via_raytracing](https://github.com/spencer-project/spencer_people_tracking/blob/master/tracking/people/spencer_tracking_utils/src/spencer_tracking_utils/simulate_occluded_detections_via_raytracing.cpp)

framarbon, 2017-11-28:
Thanks for the answers! I will have a look at the SPENCER node and also how could it be simulated in Gazebo.

framarbon, 2017-11-28:
@tlind Because obstacles are represented as aggregation of cells, it could be a workaround to publish as "input_detections" both tracked persons and obstacle cells that are in the local zone of the robot in order to take into account occlusions of static obstacles. What do you think?

RGring, 2018-09-13:
@framarbon Did you create a workaround for sensors, that consider occlusions? I need a more realistic sensor output with occlusions. Maybe you can share your progress with me?

RGring, 2018-09-17:
Okay. I will implement it.

closed 2020-06-18 by makokal