srl-freiburg/pedsim_ros #38: Create original scenario

syuntoku14, 2019-11-13:
Is there any way to create new scenario xml files? 
It seems that they are generated by SGDiCoP but I can't google it. What is SGDiCoP?

kivrakh, 2019-11-19:
Hi,

You may have a look at this to get scenario definition: [http://pedsim.silmaril.org/documentation/libpedsim/latest/demoapp_scenario.html](url)

makokal, 2019-11-30:
Hi @syuntoku14, SGDiCoP was small internal project to build a QT editor for the scenarios that we never got finished. Been meaning to work on a new python based one and would happily accept contributions along these lines

kivrakh, 2019-11-30:
> Hi @syuntoku14, SGDiCoP was small internal project to build a QT editor for the scenarios that we never got finished. Been meaning to work on a new python based one and would happily accept contributions along these lines

A practical tool would be amazing to create different types of scenarios. Otherwise, I draw the scenarios like an architect and calculate all the geometry and transform that to the XML file. How can we collaborate on this tool?

makokal, 2019-12-02:
> A practical tool would be amazing to create different types of scenarios. Otherwise, I draw the scenarios like an architect and calculate all the geometry and transform that to the XML file. How can [...]

The easiest thing is some simple python GUI that allows drag and drop and simple validation of the xml. Qt seems like a good choice for the GUI backend. 

For the way forward, I would suggest that if one of you want to take on this, please create a new implementation issue and link these discussions there. Break it down into a simple design doc and the implementation.

fverdoja, 2020-11-03:
Hi everyone and thanks for the very nice project.
One tool I could see being very useful could be one to convert from ros SLAM maps (pgm + yaml) to scenarios. If nothing of sort is already available, I could definitively work on it myself.

makokal, 2020-11-04:
@fverdoja That would be very beneficial. Please submit a PR

fverdoja, 2020-11-04:
I will put myself on it 👍

fverdoja, 2020-11-16:
@makokal I have written the first version of the script (will share it soon). 
However I am having the problem that the obstacles have default width of one meter and they are huge compared to the map (which should be already in the correct scale).
[image]
[image]
Is there a way to define a different width for an obstacle?

Thanks!

makokal, 2020-11-16:
Hi @fverdoja Thanks for taking on this. you can certainly reduce the size of the markers. These are set in https://github.com/srl-freiburg/pedsim_ros/blob/master/pedsim_visualizer/src/sim_visualizer.cpp#L229

fverdoja, 2020-11-16:
@makokal thanks! I didn't realize it was just a visualization issue. Silly me... :)
The code to build the new scenarios from ROS maps now seems to work well then. As you saw from my previous comment I have at the moment built it as a separate package from this one, but I would have no problem in integrating it in this package if you think it makes sense. 
Let me know what you prefer.

makokal, 2020-11-16:
Fantastic. Thanks for this contribution, very helpful. Like you said we can integrate it to be a sub-folder of this repo or keep it outside and add a link. I am fine with either, leave it up to you. I would ask though that at very least have a link (e.g. via a git submodule in this repo) for people to easily discover this utility, and add a few lines in the readme to point to that.

fverdoja, 2020-11-16:
I will try to send a pull request to integrate my repo into yours, I think it's probably the best approach.
Also btw, while at the moment this works with ROS maps, it is very little extra work to makit possible for it to convert any manually-drawn binary map image into scenarios. I can work on that as well. When I am done, you can probably close this issue.

fverdoja, 2020-11-23:
> Hi @fverdoja Thanks for taking on this. you can certainly reduce the size of the markers. These are set in https://github.com/srl-freiburg/pedsim_ros/blob/master/pedsim_visualizer/src/sim_visualizer.cpp#L229
@makokal regarding this, while setting up scenarios for use with ROS maps, we need 
```
walls_marker.scale.x = 0.05; 
walls_marker.scale.y = 0.05;
```
should we make these as ROS parameters so they can be set in the launchfile depending on the scenario? Ideally this would be a property of the scenario, but I am not sure we want to change the scenario definition file from the default pedsim one.

makokal, 2020-11-24:
@fverdoja Yes, making them parameters is the way to go

fverdoja, 2020-11-24:
Perfect! I will include that change in the pull request as well then.