srl-freiburg/pedsim_ros #45: Actor poses not updating in gazebo

mphamhung, 2020-01-26:
I'm not sure what I'm missing, but the actor poses aren't updating in gazebo. I'm using ubuntu16.04, ros kinetic + gazebo9. 

In order to bypass the "fatal error: pedsim_msgs/TrackedPersons.h: No such file or directory" issue during catkin_make, I create new CMakeLists.txt and package.xml files by calling:

catkin_create_pkg pedsim_gazebo_plugin gazebo_ros geometry_msgs roscpp rospy gazebo_msgs pedsim_msgs

and copying the contents to the new directory.

After building with catkin_make I run the following:
$ roslaunch pedsim_gazebo_plugin social_contexts.launch
$ roslaunch pedsim_simulator simple_pedestrians.launch 

The rviz visualization shows the actors moving, but in gazebo, the actors spawn (as grey boxes) but do not move.

closed 2020-01-27 by mphamhung

sasilva1998, 2024-03-04:
Just in case, check that the actors plugin has been added in the world file.

zerosansan, 2024-03-04:
I just checked, looks like it's already there.

social_contexts.world file:
`<plugin name="ActorPosesPlugin" filename="libActorPosesPlugin.so"> </plugin>`

zerosansan, 2024-03-14:
I managed to fix the issue. It looks like Gazebo did not find the `libActorPosesPlugin.so` when I ran with verbose.

`[Err] [Plugin.hh:180] Failed to load plugin libActorPosesPlugin.so: libActorPosesPlugin.so: cannot open shared object file: No such file or directory
`

So, the fix was to compile the `libActorPosesPlugin` custom plugin in the pedsim_gazebo_plugin folder following this tutorial (https://classic.gazebosim.org/tutorials?tut=plugins_hello_world). Then, specify the plugin path in `~/.bashrc` file as such:

`export GAZEBO_PLUGIN_PATH=:/home/zero/catkin_ws/src/pedsim_ros/pedsim_gazebo_plugin/build/devel/lib`

Now the human models are moving around :)