srl-freiburg/pedsim_ros #46: Is the robot not included in the social force model?

zzuxzt, 2020-03-17:
When I added a turtlebot to the gazebo world generated by pedsim_ros, I found that these people were not able to see the robot and collide with the robot. The pedsim_simulator has known the position of the robot in the gazebo world.

jginesclavero, 2020-03-31:
Hi @zzuxzt! I just discover how to include the robot in the social force model. You must change the "robot_mode" param by "0" in the launcher. 
In addition to that, pedsim_simulator uses the TF Tree to know the position of the robot in the global_frame.

zzuxzt, 2020-04-01:
@jginesclavero Thank you for you help. I've solved it.

takanotume24, 2020-04-22:
Hi there.
I too would like to add Turtlebot3 to Gazebo.
If you don't mind, could you give me some specifics on how to do that?