srl-freiburg/pedsim_ros #47: Human model in gazebo

FooKii, 2020-03-25:
Hello,
 I've successfully spawned agents in gazebo where agents now are in shape of box. So i'm trying to make it look like a real human. Following attempts has been made but no success: 
 
In actor_model.sdf file
- use the comment-out lines, but i see no ```model://pedsim_gazebo_plugin/meshes/actor_model.obj``` in my workspace
- use other mesh file like < walking person > from gazebo, but either i got  error ```GazeboRosApiPlugin SpawnModel Failure: input xml format not recognized```  
or the gazebo will be freezed for a moment then still no human shown in gazebo, even though in terminal i see the ```successfully spawned actor ```message

Would be great if anyone can give me a hint, 
thx in advance

BogdanLazarescu, 2020-04-09:
I would be interested in that as well :)

takanotume24, 2020-04-22:
Apparently the ``<actor>`` tag can be used to easily display an animated human being. ``<actor>`` tag is like an extension of ``<model>``, but on the sdf file it is in the same hierarchy and cannot be called.
When I call it up, the warning
```
[ WARN] [1587553447.937071639, 0.313000000]:  Could not find <model> or <world> element in sdf, so name and initial position cannot be applied
``` 
comes out.
Is it a limitation of the ROS that the ``<actor>`` tag cannot be spawned?

panzhangchao, 2020-11-20:
Have you solved the problem yet? I also wanted to include pedestrians in Gazebo, and there were problems with actor walk.dae.

bguplp, 2022-03-28:
I took human.dae from here: https://data.nvision2.eecs.yorku.ca/3DGEMS/
Used the template of actor_model.sdf of: https://github.com/coolcat647/socially-aware-walker/tree/master/catkin_ws/src/simulation/pedsim_ros/pedsim_gazebo_plugin/models

And changed the z pose to -0.75.
It makes the people move as expected but they do not move their feet.

mikhaelm321, 2024-03-12:
did anyone have solve the problem yet?