srl-freiburg/pedsim_ros #49: Question: Model parameters

bbrito, 2020-04-08:
Dear pedsim_ros team,

Thanks for this amazing simulator. Which parameters should I change in order to ensure that the pedestrians keep a larger distance among each other?

Best,
Bruno

makokal, 2020-04-13:
The separation distance is computed based on the forces acting on the agents. We do not have this nicely parametrized right now but it is possible to make it so. A starting point is the `move` method here https://github.com/srl-freiburg/pedsim_ros/blob/master/pedsim_simulator/src/element/agent.cpp#L179

The forces are weighted by the values here https://github.com/srl-freiburg/pedsim_ros/blob/master/pedsim_simulator/src/config.cpp#L37

You can tweak these to get to what you need.
If you have some spare cycles, it would be nice to add these to YAML for easy config updates

bbrito, 2020-04-13:
Nice! Thanks! I can do that. I will submit a pull request later.

Best,
Bruno