srl-freiburg/pedsim_ros #51: Pedestrians shake up and down in pedsim_gazebo_plugin

takanotume24, 2020-04-21:
Hello.
When I was using https://github.com/srl-freiburg/pedsim_ros/tree/master/pedsim_gazebo_plugin, there was a bug-like behavior, so I report it.

Watch the video on youtube, the pedestrians are swaying finely up and down. https://youtu.be/iV5yspXesCk
I'm not familiar with Gazebo so I'm not sure, but it looks like the pedestrian and the floor are physically interfering with each other.
I want to do a simulation with Gazebo and I want to solve this, do you have any solution?  
Here's what I've done
```
takuya@takuya-HP-ZBook-Studio-G3:~/catkin_ws/src/pedsim_ros$ rosrun pedsim_gazebo_plugin pedsim_to_gazbo_world.py 
>> enter pedsim scenaria name: file_name.xml 
scene1.xml
gazbo world has been generated: /home/takuya/catkin_ws/src/pedsim_ros/pedsim_gazebo_plugin/worlds/scene1.world
launch file has been generated: /home/takuya/catkin_ws/src/pedsim_ros/pedsim_gazebo_plugin/launch/scene1.launch
>> after you launch the scenario using pedsim_simulator, launch the generated world using: 
 " $roslaunch pedsim_gazebo_plugin scene1.launch"  
takuya@takuya-HP-ZBook-Studio-G3:~/catkin_ws/src/pedsim_ros/pedsim_simulator$ roslaunch pedsim_simulator airport_demo.launch airport:=scene1
takuya@takuya-HP-ZBook-Studio-G3:~/catkin_ws/src/pedsim_ros$ roslaunch pedsim_gazebo_plugin scene1.launch 

```
Thank you.

makokal, 2020-04-21:
Could this be an issue with gravity and physics in Gazebo, we haven't seen shaking outside of gazebo

takanotume24, 2020-04-22:
I also think they are the cause of this problem. Even in my environment, it is displayed normally in rviz.
Doesn't this happen in your environment?

makokal, 2020-04-22:
I am not able to test gazebo at the moment

makokal, 2020-04-28:
Thanks for the PR

closed 2020-04-28 by makokal