srl-freiburg/pedsim_ros #55: Agents Not Avoinding Robot

owen7900, 2020-05-13:
I am running the melodic version on ubuntu 18.04, using a modified version of small_airport.xml (modified by dividing all sizes and locations by 5). The agents seem to be avoiding the robot at the location that the robot spawns however they are not continuing to avoid the robot as it moves. 

Tf is configured correctly and the /robot_position topic reports the correct position, however the agents are still avoiding the (0,0) location that the robot spawns.

takanotume24, 2020-05-13:
Hi, according to #46, ``robot_model`` should be set to ``0``. It seems to work with this value in my environment.

owen7900, 2020-05-13:
Thank you for the tip. It is now crashing frequently, but I'll start a new issue for that

kajep09, 2020-05-14:
Hi @owen7900 and @takanotume24  

In #46 they mention to put the robot_mode to 0 instead of 1, but according to my tests, in both mode 0 and mode 1 the agents sort of avoid the robot, while in mode 2 they disregard it completely. See these 3 comparison videos.
[robot_mode 0](https://www.youtube.com/watch?v=DLFcZ6oBySQ&feature=youtu.be)
[robot_mode 1](https://www.youtube.com/watch?v=mixiamqSDuI&feature=youtu.be)
[robot_mode 2](https://www.youtube.com/watch?v=zFrjR5egrBY&feature=youtu.be)

However, even though the agents move around the robot to a small degree, it is not enough to stop the gazebo objects from colliding and it is also way too close to the robot if you want to test navigation etc. with move_base.

So do you know how you can increase the distance they keep from the robot? Maybe @makokal or @mahmoud-a-ali  can help here?

I assume my setup works since the agents seem to be reacting to the placement of my robot (base_footprint tf frame) so is there anything else I can change to make the agents stay further from the robot? I have already tried modifying the agent radius for the robot and a few other things, but I am not familiar enough with the code to know where to look. 
I am running these launch files along with my own robot spawned into Gazebo: 
`roslaunch pedsim_gazebo_plugin social_contexts.launch` and ` roslaunch pedsim_simulator simple_pedestrians.launch`

I am running with `with_robot` set to `false` so without 
```  <!-- Robot controller (optional) -->
  <group ns="pedbot/control" if="$(arg with_robot)">
    <include file="$(find pedsim_simulator)/launch/robot_controller.launch">
```

Does this influence anything on the agent side?

Is the robot included as an obstacle so the `force_obstacle` and `sigma_obstacle` parameters should influence this behavior or are the obstacles only for the walls? If I echo the topic `/pedsim_simulator/simulated_agents` and look at the obstacle_force for individual agents that I can see are close to the robot there is no change in the amplitude of the `obstacle_force`. It is always a very very small number (e.g. e-20 to e-40) and basically only the `social_force` and `desired_force` have any amplitude.

My temporary fix has been to modify the collisions of the agents and my robot so they don't share the same physical space in gazebo and can pass through each other. But for the purposes of testing out navigation I would still prefer much more avoidance of the robot by the agents.

makokal, 2020-05-15:
The avoidance behavior is emergent from the sum total of all the forces computed. The key variable you want to manipulate here is the agent radius, specifically, for the robot agents. Here is a starting point https://github.com/srl-freiburg/pedsim_ros/blob/master/pedsim_simulator/src/element/agent.cpp#L178

The ideal solution is to expose this robot radius so that it can be set on a launch file.

kajep09, 2020-05-16:
Hi @makokal 
Thanks for the confirmation on the agent behavior. Unfortunately, I had already tried increasing the robot agent radius to larger values without any apparent change in other agents behavior.  Maybe I was not extreme enough in choice of values - I tried 2.4m instead of 0.4m.
I also tried to set the robot velocity to some high fixed value in case that the agents would react the 
predicted position of the robot based on velocity, but didn't really try much in this regard.
I will try a little more with different radius values. 

Is the obstacle_force only for fixed obstacles such as walls or does the robot agent also modify/generate this force? Or should I only look at the social force when confirming whether the changes had any effect? I've tried following a single agent and looking at the `/pedsim_simulator/simulated_agents` topic for that agents forces, but am unsure which forces I should be looking at.

Can you briefly describe what the difference between RobotMode::CONTROLLED and RobotMode::TELEOPERATION is? At https://github.com/srl-freiburg/pedsim_ros/blob/master/pedsim_simulator/src/element/agent.cpp#L153 and forwards I am not entirely sure what is meant in terms of making the robot move and the things going on with the robot velocity. Isn't the robot position taken from the tf tree in both cases as seen in Simulator::updateRobotPositionFromTF ?

makokal, 2020-05-28:
@kajep09  Sorry I am late with this 
The modes were introduced to allow the following:
- CONTROLLED implies that the robot's velocity is set by some planning algorithm like DWA. Its very similar to the teleoperation mode
- TELEOPERATION , we map the output of a teleop node (like joystick) to drive the robot
- SOCIAL mode means the robot's velocity is set by the social force just like the other agents

makokal, 2020-06-18:
Closing this as there is no further question

closed 2020-06-18 by makokal