srl-freiburg/pedsim_ros #57: Pedsim Crashing Every Time it is run

owen7900, 2020-05-13:
I am runing Ubuntu 18.04 with Melodic and the pedsim_ros master branch. I am running with a robot enabled using a modified small_airport scenario with 5 times less people and 10 times less in all dimensions. 

[`pedsim_simulator-1] process has died [pid 136, exit code -11 `

The above line of text was spit out along with this warning several times before crashing
`[ WARN] [1589400517.323722476, 80.592000000]: Covariance matrix is not positive (semi-)definite in ProbabilityEllipseCovarianceVisual::setMeanCovariance(): Matrix3(0, 0, 0, 0, 0, 0, 0, 0, 0)`

makokal, 2020-05-15:
This is probably just a crash of the visualizer, can you run without visualization and check that the main engine works. If yes, can you share more details of the scenario you are running

owen7900, 2020-05-19:
With visualization disabled it is still crashing after about a minute attached is the scenario file (renamed to txt). The simulation runs and all agents move around properly until it abruptly crashes.
[tiny_airport.txt](https://github.com/srl-freiburg/pedsim_ros/files/4651287/tiny_airport.txt)

Any assistance would be much appreciated.
 of simulation

makokal, 2020-05-21:
Taking a look

makokal, 2020-06-18:
I tested the scenario, and it seems like the walls are too close. And be careful to spawn agents within the limits of the world