srl-freiburg/pedsim_ros #58: Robot is not moving in the rviz

chocolatebrown, 2020-05-14:
Hi,
I want to know how to enable the robot and also I can see the marker but it's not moving in the simulator. So, if there any documentation or a pointer for how to use the robot to follow a different target location or in the simulator would be helpful. I'm new to the ROS and I'm learning, so if you can point me in the right direction would be helpful for my project.

makokal, 2020-05-15:
Can you provide more information, which launch file are you using

chocolatebrown, 2020-05-18:
I was using src/pedsim_ros/pedsim_simulator/launch/simple_pedestrians.launch. control via the keyboard teleop is not working , only rqt_teleop is working.
and also how to visualize the robot other than robot frame

makokal, 2020-06-17:
The teleop stuff have been fixed now. Closing this issue.

closed 2020-06-17 by makokal