srl-freiburg/pedsim_ros #59: How to restart simulation

mphamhung, 2020-05-27:
Hi, are there any exposed functions that would let me restart the simulation without killing and relaunching? I'm looking to implementing an openaigym-like interface and want to restart the simulation for each episode.

makokal, 2020-05-30:
There is nothing exposed right now, but it should be easy to clear the scene and re-spawn the agents

mphamhung, 2020-06-01:
In simulator_node.cpp, I add the following calls too a ros listener callback:

```
      sm.cleanUp();
      sm.initializeSimulation();
      sm.runSimulation();

```

Where sm.cleanUp is defined in simulator.cpp as: 

```
void Simulator::cleanUp() {
  SCENE.clear();
}
```

This seems to get the simulator to reset on callback, except the pedsim simulator node crashes after a few times. I get an exit code -11, which sounds to me like a memory reference problem. Furthermore, each time new agents spawn, their number in rviz also increases?

Is there a different way to clear and respawn?

makokal, 2020-06-17:
Yes, this is a little more involved. We need to also reset agent ids in the libpedsim part.

NamTruongTran, 2023-09-07:
> Yes, this is a little more involved. We need to also reset agent ids in the libpedsim part. 

Hey Makokal,

Is there any solution so far?