srl-freiburg/pedsim_ros #60: Trouble with only using Gazebo but remove the RVIZ

popo1997, 2020-07-07:
I am trying to use only Gazebo for both simulation and visualization, and I hope to remove the Rviz since it is not intuitive creating an ideal scenario. Now I've succeed to launch only Gazebo during the process, but ped-sim simulator published the position due to the Rviz scenario. Since the Gazebo and Rviz have different display patterns in reading the same scenario file. What suggestions to remove Rviz and use Gazebo for the purposes. Thanks!
[image]

popo1997, 2020-07-08:
Also, when the scenario's waypoints have the parameter b(that agent will disappear at the destination), Gazebo's agent model still appears in the simulation and all blocked at the destination.Any suggestions to deal with the problem? THANKS FOR THE AMAZING PROJECT WORKS AND EFFORTS.

mikhaelm321, 2024-03-14:
@popo1997 did you find any solutions?