srl-freiburg/pedsim_ros #61: How to use pedsim?

aghorbani96, 2020-08-19:
I'm researching pedestrian dynamics and I'm new with ROS & Rviz, when I execute the launch file(e.g simple_pedestrians.launch) I encounter the empty Rviz window with no object. I would be grateful if you help me with how to use pedsim. (preferably with a sample example)
 
Thanks

[image]

aghorbani96, 2020-08-20:
Error in cmd when launching simple_pedestrians.launch:

PARAMETERS
 * /pedbot/control/driving_controller/pose_initial_theta: 0.7
 * /pedbot/control/driving_controller/pose_initial_x: 5.0
 * /pedbot/control/driving_controller/pose_initial_y: 5.0
 * /pedsim_simulator/default_queue_size: 10
 * /pedsim_simulator/enable_groups: True
 * /pedsim_simulator/max_robot_speed: 1.5
 * /pedsim_simulator/robot_mode: 1
 * /pedsim_simulator/scene_file: C:\pedworkspace\s...
 * /pedsim_simulator/simulation_factor: 1.0
 * /pedsim_simulator/update_rate: 25.0
 * /rosdistro: melodic
 * /rosversion: 1.14.6

NODES
  /
    pedsim_simulator (pedsim_simulator/pedsim_simulator)
    pedsim_visualizer (pedsim_visualizer/pedsim_visualizer_node)
    rviz (rviz/rviz)
  /pedbot/control/
    driving_controller (pedsim_simulator/simulate_diff_drive_robot)

auto-starting new master
process[master]: started with pid [4236]
ROS_MASTER_URI=http://127.0.0.1:11311
setting /run_id to 4a67a58f-e2c6-11ea-8fe7-ac2b6e803a36
process[rosout-1]: started with pid [10032]
started core service [/rosout]
[31mERROR: cannot launch node of type [pedsim_simulator/pedsim_simulator]: Cannot locate node of type [pedsim_simulator] in package [pedsim_simulator]. Make sure file exists in package path and permission is set to executable (chmod +x)[0m
[31mERROR: cannot launch node of type [pedsim_simulator/simulate_diff_drive_robot]: Cannot locate node of type [simulate_diff_drive_robot] in package [pedsim_simulator]. Make sure file exists in package path and permission is set to executable (chmod +x)[0m
[31mERROR: cannot launch node of type [pedsim_visualizer/pedsim_visualizer_node]: Cannot locate node of type [pedsim_visualizer_node] in package [pedsim_visualizer]. Make sure file exists in package path and permission is set to executable (chmod +x)[0m
process[rviz-5]: started with pid [11764]
[ERROR] [1597916556.850507800]: PluginlibFactory: The plugin for class 'spencer_tracking_rviz_plugin/TrackedPersons' failed to load.  Error: Could not find library corresponding to plugin spencer_tracking_rviz_plugin/TrackedPersons. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1597916556.864185400]: PluginlibFactory: The plugin for class 'spencer_tracking_rviz_plugin/TrackedGroups' failed to load.  Error: Could not find library corresponding to plugin spencer_tracking_rviz_plugin/TrackedGroups. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.

makokal, 2020-09-08:
What versions of ROS, ubuntu  are you using?

aghorbani96, 2020-09-10:
I'm using melodic for windows.

mengxingshifen1218, 2021-11-30:
hello，I have same question， how to solve？

makokal, 2021-11-30:
Can you share the output of compiling the packages successfully. It seems that you have not build all the packages