srl-freiburg/pedsim_ros #66: Running on ROS Noetic

tanyouliang95, 2020-10-19:
Hi, I was trying to run this on a fresh ROS noetic with `melodic-devel` branch. Managed to compile it, but getting this error during launching of the launch file. Would kindly need some advice on this. Thanks!

```bash
[ WARN] [1603085782.510908291]: WARNING: package://spencer_tracking_rviz_plugin/media/animated_walking_man.mesh is an older format ([MeshSerializer_v1.41]); you should upgrade it as soon as possible using the OgreMeshUpgrade tool.
[ WARN] [1603085782.511277907]: Can't assign material male_base_lambert1 to SubEntity of mesh_person_visual0 because this Material does not exist. Have you forgotten to define it in a .material script?
[ WARN] [1603085782.516801097]: OGRE EXCEPTION(5:ItemIdentityException): Cannot find a group named rviz in ResourceGroupManager::isResourceGroupInitialised at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 1916)
terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what():  OGRE EXCEPTION(5:ItemIdentityException): Cannot find a group named rviz in ResourceGroupManager::isResourceGroupInitialised at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 1916)
[rviz-5] process has died [pid 14458, exit code -6, cmd /opt/ros/noetic/lib/rviz/rviz -d 
```

yash-goel, 2020-11-06:
Hi, I am using ROS noetic as well. Instead of `melodic-devel`, try the latest branch once. That worked for me. The error you have seems to be because of old mesh format - possibly due to `melodic` branch - which is quite behind the latest one.

tanyouliang95, 2020-11-08:
Thanks for the reply. Running well now :+1:

closed 2020-11-08 by tanyouliang95

mincheulkim, 2022-09-14:
Could you tell me what is the latest branch to solve the OGRE issue?