srl-freiburg/pedsim_ros #67: Defining an agent with `n="1"` spawns no agent instead

fverdoja, 2020-11-25:
Hi,
I was trying to create a scenario where a single agent was defined.
Something like:
```
<?xml version="1.0" ?>
<scenario>
  <!-- some obstacles here -->
  
  <waypoint id="w1" r="0.2" x="18" y="9"/>
  <waypoint id="w2" r="0.2" x="20" y="8"/>
  <waypoint id="w3" r="0.2" x="21" y="6"/>
  
  <agent dx="0.5" dy="0.5" n="1" x="18" y="7">
    <addwaypoint id="w1"/>
    <addwaypoint id="w2"/>
    <addwaypoint id="w3"/>
  </agent>
</scenario>
```
However, having agent with property `n="1"` doesn't actually spawn any agent. None is shown in rviz, and no pose is published in the corresponding topic. I tried also omitting the `n` property, or setting it to `n="0"` and `n="-1"` but it achieves the same effect. I tried also playing around with `dx` and `dy` to no avail.

Is this a bug?
From the [pedsim documentation](http://pedsim.silmaril.org/documentation/libpedsim/latest/demoapp_scenario.html) it seems it should be possible:

> Since it would not be comfortable to define each of potentially many agents individually, there is a way to specify groups of agents. In the agent definition, a agent multiplier `n` can be added. _N_ copies of that agent will be spawned into the simulation.

If it is working as intended, is there any way to define a single agent instead of a group?

kivrakh, 2021-02-01:
I think you forget to add the **type** parameter in agent tag
` <agent dx="0.5" dy="0.5" n="1" x="18" y="7" type="1">`

Also you should add the robot to the scene by adding type="2". Here is the scene file I have created before:
```
<?xml version="1.0" encoding="UTF-8"?>
<scenario>
    <!--Obstacles-->
    <waypoint id="start1" x="10" y="4" />
    <waypoint id="goal1" x="10" y="4" />

    <agent x="10" y="4" n="1" dx="0" dy="0" type="1">
        <addwaypoint id="goal1"/>
        <addwaypoint id="start1"/>
    </agent>

    <waypoint id="start8" x="40" y="1" />
    <waypoint id="goal8" x="4" y="1" />
    <!--robot-->
    <agent x="1" y="4" n="1" dx="0" dy="0" type="2">
        <addwaypoint id="goal8"/>
        <addwaypoint id="start8"/>
    </agent>
</scenario>
```

fverdoja, 2021-02-05:
Hi, thanks for your answer.
Thanks for pointing out I was missing the `type` tag.

The example you provided works, however, if I remove any of the two agents (so only the robot or only the human is left) no agent gets spawn, regardless of if they have `type` tag 1 or 2, or no `type` tag at all.

Is that an intended behaviour?

fagnerpimentel, 2021-03-18:
I have the same problem.

kivrakh, 2021-03-19:
Hello @fverdoja and @fagnerpimentel,

As far as I experience, the robot is spawned even you remove any of the two agent. If you can give precise information about what you want to do then maybe we can solve it.

fverdoja, 2021-03-22:
Hi, I am trying to run the simulator without the robot present to record people movement tracks. I would like to be able to spawn different amounts of (non robot) agents each time, but when I try to spawn one, no agent is spawn instead.

fagnerpimentel, 2021-03-23:
Hi, @kivrakh thank you for the help. I just put a type 2 agent in my XML scenario as you show.
`  <agent x="0" y="0" n="1" dx="0" dy="0" type="2"/>`

But I agree with the @fverdoja. This may be not the intended behavior.

This works: (1 person and 1 robot)
```
<scenario>
  <agent x="0" y="0" n="1" dx="0" dy="0" type="2"/>
  <agent dx="2" dy="2" n="1" x="0" y="0" type="0"/>
</scenario>
```

This works: (2 persons)
```
<scenario>
  <agent dx="2" dy="2" n="2" x="0" y="0" type="0"/>
</scenario>
```

This not works: (1 person)
```
<scenario>
  <agent dx="2" dy="2" n="1" x="0" y="0" type="0"/>
</scenario>
```

fagnerpimentel, 2021-03-25:
Hi guys, 
I think I found the problem.

In this file, at lines 291-293:
https://github.com/srl-freiburg/pedsim_ros/blob/master/pedsim_simulator/src/simulator.cpp

there is this condition:
```
  if (SCENE.getAgents().size() < 2) {
    return;
  }
```
I don't know why this is there but seems there is no problem if you remove it since this function is just to publish the agents.

Just comment or delete these lines, run catkin_make and it is done.
It should work for only one person now.
```
<scenario>
  <agent dx="2" dy="2" n="1" x="0" y="0" type="0"/>
</scenario>
```

kivrakh, 2021-03-31:
Yep, that works @fagnerpimentel. Nice job!