srl-freiburg/pedsim_ros #71: Social drive robot mode not working

sasilva1998, 2021-04-13:
I am assigning the robot the following configuration with waypoints with `robot_mode` equal to 2 but it does not move according to the social forces model, does anyone have an idea why?

````
<!--robot-->
  <agent x="0" y="0" n="1" dx="0" dy="0" type="2">
    <addwaypoint id="office4"/>
    <addwaypoint id="office3"/>
  </agent>
````

ThaiLe79, 2021-07-07:
> I am assigning the robot the following configuration with waypoints with `robot_mode` equal to 2 but it does not move according to the social forces model, does anyone have an idea why?
> 
> ```
> 
>   <agent x="0" y="0" n="1" dx="0" dy="0" type="2">
>     <addwaypoint id="office4"/>
>     <addwaypoint id="office3"/>
>   </agent>
> ```

Have you done this yet? I am also having the same problem

sasilva1998, 2021-07-08:
i haven't been able to solve it, for my purpose I just started to write my own python code to calculate the forces and control a robot I have.

ThaiLe79, 2021-07-24:
I see in there topic /pedsim_simulator/robot_position. I thought it would be possible to write a node to use the information in this topic to control the robot I have. Do you do the same?

ThaiLe79, 2021-07-28:
I have downloaded the package that contains the node you said. But I can't
execute. Can you help me? I really need that. Thank you so much.