srl-freiburg/pedsim_ros #75: Rviz Crashes (OGRE)

IhabMohamed, 2021-08-25:
Hello Everyone,

I got the following error when I run the `airport_demo` example (**roslaunch pedsim_simulator airport_demo.launch**). So, I am not able to visualize the crowds and moving people in `rviz`. 

I am using ROS `noetic` and `Ubuntu 20.04`.

Error:
`OGRE EXCEPTION(5:ItemIdentityException): Cannot find a group named rviz in ResourceGroupManager::isResourceGroupInitialised at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 1916)`

I tried to reinstall `rviz` and `libogre-1.9-dev` [(see this LINK](https://answers.ros.org/question/195684/rviz-crashes-ogre/)

Thanks in advance for your help.

maprdhm, 2022-01-21:
Hello,

I had the same error on ROS noetic.
After reading [this discussion](https://github.com/locusrobotics/robot_navigation/issues/86),  I replaced the line:
`Ogre::MaterialPtr default_material = Ogre::MaterialManager::getSingleton().create(ss.str(), "rviz");`
by 
`Ogre::MaterialPtr default_material = Ogre::MaterialManager::getSingleton().create(ss.str(), Ogre::ResourceGroupManager::DEFAULT_RESOURCE_GROUP_NAME);`
in the files 2ndparty/spencer_tracking_rviz_plugin/src/visuals/mesh_node.h and 2ndparty/spencer_tracking_rviz_plugin/src/visuals/person_visual.cpp,

And the error did not appear anymore.
Hope this helps, even if the issue is old...

IhabMohamed, 2022-01-31:
Thank you so much, @maprdhm. I will follow your suggestions, then I will keep you updated.

IhabMohamed, 2022-02-04:
It works well, thanks @maprdhm

closed 2022-02-04 by IhabMohamed