srl-freiburg/pedsim_ros #77: Possible segfault culprit

ksatyaki, 2022-04-21:
https://github.com/srl-freiburg/pedsim_ros/blob/de3772630ae174b278da00bcf26e5cf6c2db5ab2/pedsim_simulator/src/scene.cpp#L260

Perhaps needs a continue after deletion? Because `currentGroup` is accessed immediately afterwards. It's bound to fail, right? Or am I missing something?

tlind, 2022-04-21:
Where do you think it is accessed immediately afterwards?
The expression in line 261 and the statement in line 263 are inside an "else" block. Thus they cannot be executed at the same time in which the delete statement (inside the if that belongs to this else) is invoked, as they are in alternative code paths.

ksatyaki, 2022-04-21:
I saw that someone else had ported most of `pedsim_ros` to ROS2. I am completing just that. :-)