srl-freiburg/pedsim_ros #80: catkin_make error

InguChoi, 2022-08-05:
[image]
[image]

Do you know this error?
I just build this repository...

zerosansan, 2022-10-01:
what version of ROS are you using?

Namo98, 2023-08-05:
> what version of ROS are you using?
I got the same error and I using noetic as distribution do you have any solution for that?
(I cloned the master branch of pedsim)

Thank you very much in advance.

s0um0r0y, 2025-05-09:
I am facing a similar issue with pedsim_simulator file here I am attaching a screenshot below for reference

[image]