srl-freiburg/pedsim_ros #81: crash when style change to 'Person meshes' in RViz

mincheulkim, 2022-09-14:
Hi, I finally installed latest repo in Noetic, and ran simple_pedestrains.launch. Rviz is successfully run, but when I changed TrakcedPersons-Style to 'Personal meshes', the program is crashed.

[image]

closed 2022-09-14 by mincheulkim

Namo98, 2023-08-05:
Hey Mi

> Hi, I finally installed latest repo in Noetic, and ran simple_pedestrains.launch. Rviz is successfully run, but when I changed TrakcedPersons-Style to 'Personal meshes', the program is crashed.
> 
> [image]

Hey mincheulkim,
did you find a solution? 
I have the same error message.
Thank you so much ina advance

mikhaelm321, 2024-03-12:
@mincheulkim hey could you tell me how you solve it?