srl-freiburg/pedsim_ros #84: ROS2 Humble branch

stephenadhi, 2022-11-18:
Hi! I'm migrating pedsim_ros to ROS2 and looking for people also to contribute. I forked the latest ROS2 PR as the base repo (unfortunately as of now, they are not working out of the box anymore in the newest distro... not even in galactic or foxy is working as well). Since the galactic is reaching the end of life and foxy will be out of support next year, could you open a humble branch in the repo, please?

Thank you so much.

Namo98, 2023-08-04:
Hey Stephen,

thank you for your pedsim_ros ros2 repository. 
I want to use foxy as ros distribution instead of humble.
After i made some changes (e.g. change tf2_geometry_msgs/tf2_geometry_msgs.hpp to tf2_geometry_msgs/tf2_geometry_msgs.h etc.), colcon build builded successfully without any errors:

After i run:
ros2 launch pedsim_simulator house_demo_launch.py

I getting the following message repeatly:
[pedsim_simulator-1] [WARN] [1691145349.541933382] [pedsim_simulator]: "odom" passed to lookupTransform argument target_frame does not exist. 
[pedsim_simulator-1] [WARN] [1691145354.581924814] [pedsim_simulator]: "odom" passed to lookupTransform argument target_frame does not exist.

Do you have any solution for that.

Thank you very much in advance.

stephenadhi, 2023-08-04:
Hey, you're welcome. You need to have your own robot running that has the odom frame.

Namo98, 2023-08-06:
Hey Stephen, 

first thank you very much for you reply. I really appreciate your time. 
I thought you house_demo_launch.py file is a example. 
How can i use my own robot maybe with turtlebot3 simulation?
I tested the pedsim ros1 there was the simple_pedestrians.launch file. 
I want to test the pedsim on ros2 too. I didn't find any approaches so far. 

Thank you so much !

Namo98, 2023-08-06:
> Hey, you're welcome. You need to have your own robot running that has the odom frame.

Hey Stephen,

first thank you very much for you reply. I really appreciate your time.
I thought you house_demo_launch.py file is a example.
How can i use my own robot maybe with turtlebot3 simulation?
I tested the pedsim ros1 there was the simple_pedestrians.launch file.
I want to test the pedsim on ros2 too. I didn't find any approaches so far.

Thank you so much !

stephenadhi, 2023-08-07:
Hey @Namo98 ,

happy to help! The test launch is now ready, please refer to https://github.com/srl-freiburg/pedsim_ros/pull/86#issuecomment-1668151892. For further things please open an issue in the forked repository itself. Closing this now.

closed 2023-08-07 by stephenadhi