srl-freiburg/pedsim_ros #92: Rviz Crashes When Changing 'Style' under 'TrackedPersons' to 'Person meshes'

HazelFannn, 2024-01-11:
Hello,

I'm using Ubuntu 20.04 Melodic. After launching the simulator in Rviz, all the moving pedestrians are initially in the style of 'Bounding boxes'. I can change the style to all other options except for the 'Person meshes'.
[image]

If I select 'Person meshes', Rviz gets terminated and these error messages arise:
[image]

I would really appreciate it if anyone could help with the problem.

HazelFannn, 2024-03-12:
@mikhaelm321 Yes I did find out why. It's simply the Melodic version is not paired with Ubuntu 20.04... Try to run the simulation in Ubuntu 18.04 and it should work. I'd recommend using a docker container instead of directly changing your host machine's system.

HazelFannn, 2024-03-12:
Resolved

closed 2024-03-12 by HazelFannn

mikhaelm321, 2024-03-13:
> @mikhaelm321 Yes I did find out why. It's simply the Melodic version is not paired with Ubuntu 20.04... Try to run the simulation in Ubuntu 18.04 and it should work. I'd recommend using a docker container instead of directly changing your host machine's system.

@HazelFannn  yeah, I have tried it too on ubuntu 18.04 and the RVIZ work fine. But why on the gazebo the "pedestrian" dont appear as human shape? mine was appear as walking boxes. Did you  have the same problem?