srl-freiburg/pedsim_ros #95: Build fail in Noetic

Alex108306, 2024-05-12:
[image]
I have encountered this issue.

Alex108306, 2024-05-12:
some issue with file "additional_topic_subscriber.h"

Alex108306, 2024-05-12:
/home/giang/ped_ws/src/pedsim_ros/2ndparty/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’

WasFun00, 2024-05-22:
have u ever solved?

Alex108306, 2024-05-25:
Yeah I've solved it. There is an error in syntax.

nanaminh, 2024-06-13:
@Alex108306  Hello, I am having the same problem. Could you share how you solved the problem?

nanaminh, 2024-06-17:
For anyone looking for the solution later, here's how I solved the problem:

In additional_topic_subscriber.h, line 63, 
change
class _AdditionalTopicSubscriber: QObject
to
class _AdditionalTopicSubscriber: **public** QObject